rpng/R-VIO #2: Results with respect to moving world

mmmfarrell, 2019-03-21:
Hey @huaizheng ,

I'll be honest I haven't really dug through your paper or code thoroughly yet, however, I am very familiar with VINS. I was wondering if you had any results or intuition for how well your method would perform in a moving world (i.e. all the visual feature tracked belong to some moving vehicle). Does this break underlying assumptions of your approach? or since everything is focused on the relative pose is this still ok? I know VINS, ROVIO, SVO and others all assume that the features are static in the inertial frame, so these quickly break when tracking moving features.

In the meantime I will read through your paper the see if I can get an idea for myself. Awesome work though.

huaizheng, 2019-03-22:
Thanks, Michael. Here I would like to share my own experience with you about tracking moving features. First of all, VINS is usually used to estimate absolute motion of a sensing platform with respect to some global frame of reference. However, most of time what we have are only the relative measurements (e.g. from a camera) which cannot give us any global information but can provide geometric constraints for motion estimation. If we treat a (repeatedly observed) moving feature as static, then such geometric constraints will be inconsistent to the motion prediction obtained from IMU and easily cause VINS diverge. Thus, in order to recover the motion correctly, either in world-centric or in robocentric, you need to put some efforts on the moving feature rejection (e.g., using RANSAC). Or you may want to track those moving features and still use them for estimation, then a motion model will be needed to predict the feature's motion (i.e., a moving object tracking problem), however, you need to be careful with some degenerate motion cases between the VINS and the feature (e.g., moving in parallel, or moving towards each other) when motion ambiguity emerges. So, this is why most of the existing VINS algorithms assume that the features used for estimation are static. Hope those could help you somewhat.

mmmfarrell, 2019-03-22:
@huaizheng thanks for the very detailed response. I really should have read through your paper before asking the question. I though your work was directly related to [VINS-Mono](https://arxiv.org/pdf/1708.03852.pdf) and I completely misunderstood the contribution you are making. I have been working on tracking moving features on a rigid body with a known motion model and indeed it is a difficult problem. I appreciate your insight about it.

closed 2019-03-22 by mmmfarrell