rpng/R-VIO #4: The trajectory will divergent when RVIO come up with still scenes for a period of time after initiliaztion.

ChenHancheng, 2019-07-04:
如果数据集的数据在最后有一段比较长的静止的场景，rvio估计出来的轨迹会发散，我觉得可能的原因是视觉部分每次都会重新估计landmark的3D坐标，而不是像orb_slam一样，保存之前的landmark，不知道对不对？

huaizheng, 2019-07-04:
You're right. This is the main difference between SLAM and VIO. Although you may find that some VINS also call itself VIO, it is actually implemented on the SLAM framework because of jointly estimating the features.

ChenHancheng, 2019-07-04:
那您觉得除了保存3d landmark之外比较好的一个解决方案是什么呢？

huaizheng, 2019-07-04:
First of all, this actually depends on your application, because the key point is whether you need an IMU for estimation, because as compared with camera usually it is hard to handle IMU noise when it is stationary. So if you are doing structure from motion (no scale) then the visual SLAM is good enough, like the orb_slam you mentioned. While, if you want to do the navigation then the R-VIO here only provides the functionality of 3D motion tracking, and a visual-inertial SLAM system will be an ultimate solution which I am now working on.

ChenHancheng, 2019-07-05:
好的，谢谢~

closed 2019-07-05 by ChenHancheng