rpng/R-VIO #5: gravity need be normalized?

canyou2014, 2019-07-06:
Thanks for your sharing of the code. I tested it on my own datasets, I found that the vio system is   unstable, but if I remove the normalizing of gravity, it run more stably.  Should the gravity be normalized?

huaizheng, 2019-07-06:
Thank you @canyou2014 for testing my code. For the issue you met, I may think that it was not caused by the gravity. While first of all, could you please tell me where you changed in the code? Because there are several places need to be changed in order to use regular gravity estimate, I need to make sure that you did right modification.

canyou2014, 2019-07-06:
@huaizheng  I removed this [line](https://github.com/rpng/R-VIO/blob/master/src/rvio/Updater.cc#L573) . And I found in the prePrediction process,  in the first iteration, the gk is a unit vector, but after the the first iteration, the gk become a regular gravity value, the code is [here](https://github.com/rpng/R-VIO/blob/master/src/rvio/PreIntegrator.cc#L132).

huaizheng, 2019-07-06:
OK, I see. Your modification is not enough to avoid the gravity normalization. As you noticed the code in the preintegration, if you would like to use regular gravity then every places where I have multiplication by the magnitude of gravity needs to be changed, or the resulting estimation would be wrong. And the line you marked in the iteration is about one of the entries of state transition matrix and we need regular gravity value to compute it, which you may refer to our paper. So, the cause of this issue may come from some other places, for example the config file. Did you modify the values of parameters there based on the sensing platform you use?

canyou2014, 2019-07-06:
@huaizheng I also found a large z-axis drift on my own datasets,  especially on pure rotation datasets，I think it maybe caused by a drift on the estimated gravity. What's your idea? Hope for your reply.

canyou2014, 2019-07-06:
@huaizheng Yes, I remove all 'g.normalized()' from the code, and I also modified the values of parameters, including the noises of imu and image. It runs well on my own datasets (thanks for your sharing again) , but there are also some small problems, such as z-axis drift and static motion drift.

huaizheng, 2019-07-06:
Did you get rid of all the multiplications of the magnitude of gravity, e.g., https://github.com/rpng/R-VIO/blob/d8f5587c7c466fd10559c8e93765ae11f1f74d6f/src/rvio/PreIntegrator.cc#L67?

canyou2014, 2019-07-06:
@huaizheng Yes, of course I get rid of all multiplications of the magnitude of gravity, and modify all the jacobian matrix about gravity as the same as your paper refered.

huaizheng, 2019-07-06:
I see, thanks. Although this works, to be honest, I do not recommend it. Because it is hard to guarantee that the magnitude of the estimated gravity is always around 9.8, especially when using some low-price IMU, and the error in the magnitude would eventually diverge the estimator. So this is why I treat the gravity as a vector of known magnitude, and always do normalization in the code. Also, if you can use regular gravity to let it run, then definitely it should work with its original form.

canyou2014, 2019-07-06:
@huaizheng  I see, thanks.

closed 2019-07-06 by canyou2014