rpng/R-VIO #7: R-VIO initialization

chutsu, 2019-07-10:
Was wondering if R-VIO needs the IMU to be stationary for a minimum time for initialization? 

What I'm seeing after testing around 3 MAV EuRoC bags is that the bags that start with the drone stationary at the beginning seems to initialize well. However, the `MH_01_easy.bag`  dataset starts with the drone being moved up and down first (presumably to initialize a VO), and that seems to cause R-VIO to diverge significantly at the start. Could this be because the sliding window hasn't been filled yet and the up and down motion cause the intialization to fail? If so this might be worth mentioning in the README?

[image]

chutsu, 2019-07-10:
Just realized this is basically the same question as Issue #1

huaizheng, 2019-07-10:
Thanks @chutsu for testing the code. The initialization that I implemented is very simple, if you refer to the corresponding part of code. So, here I would like to clarify that current R-VIO codebase is for validation of the idea proposed in our paper and provides basic navigation functionality. If you want a full-function solution (e.g., with a better initialization), then either you could freely implement your own one or I recommend some other open-source packages for use (e.g., OKVIS or VINS-Mono).

chutsu, 2019-07-10:
>  I would like to clarify that current R-VIO codebase is for validation of the idea proposed in our paper and provides basic navigation functionality

Thats fair enough. Thanks for sharing your code! :)

closed 2019-07-10 by chutsu