rpng/R-VIO #8: Trajectory drifting for Realsense D435i camera.

guangyuzhou-wework, 2019-07-19:
Hello! I am trying to run your code on my RS D435i camera. I have already changed all the parameters in the yaml file according to my camera settings. However, when I start to run rviz, the trajectory always starts to drift largely. Currently all I am doing is to tune with different parameters. I didn't change the code at all. Could you please give me some insight? Thank you very much!

huaizheng, 2019-07-19:
Thanks @guangyuzhou-wework  for testing our code. Personally, I do not have the same camera at hand, but I just went to its website to check some relevant information. First of all, could you please tell me how you collected the data, e.g., using which kind of driver? Because some sensor may have its own build-in software to preprocess IMU data, like eliminating the gravity effect, so the accelerometer outputs would purely reflect the motion. Also, if you found that R-VIO started to drift very quickly even from stationary, then you need to check the IMU readings first to make sure that they are the raw data.

guangyuzhou-wework, 2019-07-19:
Thank you for the quick response @huaizheng ! 
1. I think I am using the D400 drivers [https://downloadcenter.intel.com/download/28870/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?product=190004](here). Then, I simply use realsense ros to publish the camera/imu topic. 
2. Here is an example reading of the IMU. I think it should be the raw data. 
`orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00708649028093
  y: 0.0244191940874
  z: 2.52911850112e-05
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -0.0280539542437
  y: -9.71505069733
  z: 0.675603687763
`Anything abnormal from above?

huaizheng, 2019-07-19:
Those values look reasonable, so next thing you may need to check is the camera-to-IMU calibration parameters in order to make sure that the transformation direction is correct, e.g., T_BC0 is used for converting a coordinate from camera frame to IMU frame.

guangyuzhou-wework, 2019-07-19:
Thanks again. Actually I have extract the extrinsic parameters by using "rs-sensor-control". I choose to view the RGB-camera to IMU frame values, which is as below. 
Translation Vector (T) : [-0.0204567,0.00497697,0.0111223]
Rotation Matrix (R)   
                  : [0.999911,0.0104414,-0.00831997]
                   : [-0.0104136,0.99994,0.00336883]
                   : [0.00835464,-0.00328189,0.99996]
Then to feed into the yaml file, I append them and assign as:
    data: [0.999911,0.0104414,-0.00831997, -0.0204567,
            -0.0104136,0.99994,0.00336883, 0.00497697,
            0.00835464,-0.00328189,0.99996, 0.0111223,
              0.0, 0.0, 0.0, 1.0] 
If you spot anything unusual, please let me know :) 
My yaml file is here: [rvio_RS1.yaml.tar.gz](https://github.com/rpng/R-VIO/files/3412370/rvio_RS1.yaml.tar.gz)

I also attached the drifting trajectory with the above parameter settings. I didn't move the sensor at all. 

[image]

huaizheng, 2019-07-19:
I see, thanks. I just checked your config file and found those issues: 1) the IMU rate looks strange and the IMU noise values are not your own sensor's which may cause big problem, 2) the camera rate also looks strange and the image noise value should be set as the inverse of focal length (1/f), and 3) in the extrinsic part you did not fully comment out the one that you did not use so the system would read wrong transformation matrix. Please remember that all the parameters need to be carefully set according to your own sensor, or you will definitely get the wrong result.

guangyuzhou-wework, 2019-07-19:
Thank you huai. After I have done the following, I still got the drifting results. 
1) I re-checked the rate of IMU and camera. Now IMU rate is set as 250Hz and camera rate as 30Hz. 
2) I measured the imu noise values of my device by using the[ imu util package ](https://github.com/gaowenliang/imu_utils), and I got:
IMU.sigma_g: 6.4487419686365066e-01
IMU.sigma_wg:  3.6514005201966904e-03
IMU.sigma_a: 2.5017156619950005e+00
IMU.sigma_wa: 1.1530558437344696e-01
3) I think the issue may stem from the extrinsic part. Could you please provide some suggestions on verifying I got the correct  transformation matrix? How may I check the order? I really appreciate your help.
Best
Guangyu

huaizheng, 2019-07-21:
Currently, my suggestion is to make sure of everything about your sensor before testing the code. The IMU parameters you provided seem not correct, especially the orders of magnitude of sigma_a and sigma_wa are too big. I suggest you to manually compute them after checking the document of IMU, for which you can refer to the procedures provided by [Kalibr](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model). For the extrinsic parameters, you may obtain either T_ic (from IMU to camera) or T_ci (from camera to IMU) from the calibration tools used, while R-VIO needs T_ic (i.e., T_BC0), so you need to make sure that the matrix in use is correct.

chutsu, 2019-07-22:
> IMU rate is set as 250Hz
@guangyuzhou-wework That doesn't sound right. The Intel D435i accelerometer is 62.5 or 250Hz, while the gyro is 200 or 400Hz. By default the Intel D435i does not output a single IMU message because both the accelerometer and gyroscope output at different rates. So I'm curious if you used their official "sync" method to interpolate both accelerometer and gyroscope. To be specific, there's a `unite_imu_method` parameter you can configure in the ros launch file, see [this](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/includes/nodelet.launch.xml#L83).

There are a couple things wrong with the Intel D435i:
1. Be warned the interpolation implementation is buggy and I don't know if the fix has been merged into devel. [See this issue](https://github.com/IntelRealSense/realsense-ros/issues/703).
2. The D435i has 1 RGB and 2 IR cameras, *DO NOT USE THE RGB* as it is rolling shutter. Only the IR stereo cam is global shutter. If you can use the IR stereo camera stream. You'll have to disable the IR Emitter if you go down that route. 
3. Last time I checked there's a time delay between IR and IMU [see here](https://github.com/IntelRealSense/realsense-ros/issues/563), so thats another problem.

MoBaT, 2019-07-22:
Hey @chutsu, nice repo that you started! Have you been able to get around the time delay between IR and IMU yet? Is it a constant delay or is it variable? Do you have any suggestions on where to start for me to diagnose the issue and get proper timestamping between IR and IMU?

chutsu, 2019-07-22:
> Have you been able to get around the time delay between IR and IMU yet?

I plan on doing that at some point this week, but in the mean time you can steal the calibration results from `andre-nguyen` (github user) in this [github issue](https://github.com/IntelRealSense/realsense-ros/issues/563) to start. He used [Kalibr](https://github.com/ethz-asl/kalibr) to obtain his results. It is arguably the best open sourced cam-imu calibrator out there at the moment. Its not easy to use but there isn't really an alternative I can think of.

```
Calibration results
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 0.173943353672, median 0.147643722052, std: 0.119621573992
Reprojection error (cam1):     mean 0.16247533893, median 0.138554530845, std: 0.10923907102
Gyroscope error (imu0):        mean 0.32260865406, median 0.252569401111, std: 0.233390189106
Accelerometer error (imu0):    mean 0.778813662592, median 0.641485282384, std: 0.72379960621

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 0.173943353672, median 0.147643722052, std: 0.119621573992
Reprojection error (cam1) [px]:     mean 0.16247533893, median 0.138554530845, std: 0.10923907102
Gyroscope error (imu0) [rad/s]:     mean 0.0134788896442, median 0.0105525845083, std: 0.00975125919095
Accelerometer error (imu0) [m/s^2]: mean 0.0324786074339, median 0.0267516476173, std: 0.0301843745173

Transformation (cam0):
-----------------------
T_ci:  (imu0 to cam0): 
[[ 0.0110382  -0.99991198  0.00736134  0.00624933]
 [-0.00558312 -0.00742331 -0.99995686 -0.00097011]
 [ 0.99992349  0.01099662 -0.00566457 -0.01621383]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam0 to imu0): 
[[ 0.0110382  -0.00558312  0.99992349  0.01613819]
 [-0.99991198 -0.00742331  0.01099662  0.00641987]
 [ 0.00736134 -0.99995686 -0.00566457 -0.00110792]
 [ 0.          0.          0.          1.        ]]

timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
-9.29860761026e-05

Transformation (cam1):
-----------------------
T_ci:  (imu0 to cam1): 
[[ 0.00805359 -0.99993973  0.00746128 -0.04341998]
 [-0.00547116 -0.00750548 -0.99995687 -0.00099582]
 [ 0.9999526   0.00801242 -0.00553128 -0.01652654]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam1 to imu0): 
[[ 0.00805359 -0.00547116  0.9999526   0.01686999]
 [-0.99993973 -0.00750548  0.00801242 -0.04329242]
 [ 0.00746128 -0.99995687 -0.00553128 -0.00076322]
 [ 0.          0.          0.          1.        ]]

timeshift cam1 to imu0: [s] (t_imu = t_cam + shift)
-0.00011329653378

Baselines:
----------
Baseline (cam0 to cam1): 
[[ 0.99999554 -0.00008307 -0.00298525 -0.04971776]
 [ 0.0000834   0.99999999  0.00011104 -0.00002442]
 [ 0.00298524 -0.00011129  0.99999554 -0.00033155]
 [ 0.          0.          0.          1.        ]]
baseline norm:  0.0497188743763 [m]

Gravity vector in target coords: [m/s^2]
[-0.0534996  -8.66839068 -4.58525503]

Calibration configuration
=========================

cam0
-----
  Camera model: pinhole
  Focal length: [373.6504302988745, 375.75763034879947]
  Principal point: [320.4363427220866, 239.39529301144844]
  Distortion model: radtan
  Distortion coefficients: [-0.004020373096063456, -0.005547787897453452, 0.0003561157710439325, 0.0003653801898754835]
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.0235 [m]
    Spacing 0.00705 [m]

cam1
-----
  Camera model: pinhole
  Focal length: [373.40636529301906, 375.49353064695407]
  Principal point: [321.520615553491, 239.36084632157363]
  Distortion model: radtan
  Distortion coefficients: [-0.0024021276332198294, -0.0078016361176795074, 0.0002687660056814919, 0.0004465628578118777]
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.0235 [m]
    Spacing 0.00705 [m]

IMU configuration
=================

IMU0:
----------------------------
  Model: calibrated
  Update rate: 500.0
  Accelerometer:
    Noise density: 0.001865 
    Noise density (discrete): 0.0417026677804 
    Random walk: 0.0002
  Gyroscope:
    Noise density: 0.0018685
    Noise density (discrete): 0.0417809301596 
    Random walk: 4e-06
  T_i_b
    [[ 1.  0.  0.  0.]
     [ 0.  1.  0.  0.]
     [ 0.  0.  1.  0.]
     [ 0.  0.  0.  1.]]
  time offset with respect to IMU0: 0.0 [s]
```

chutsu, 2019-07-22:
> Is it a constant delay or is it variable?

If it isn't constant than we're pretty screwed. I don't know of any established VIO systems out there that assume the delay is variable. Correct me if I'm wrong but most if not all assume it is constant, its mostly to do with how it adds complexity to the VIO formulation.

> Do you have any suggestions on where to start for me to diagnose the issue and get proper timestamping between IR and IMU?

Kalibr is a good start. But other than calibrating and trying it out I personally am not sure how else to verify the imu-cam ridgid body transform in a rigourous manner.

MoBaT, 2019-07-22:
Awesome. That's what I thought. I wasn't too sure what tools others using to get it done. I have used Kalibr many times before and the same with the Allan plots using a variety of cameras and IMUs. I'll calibrate the IMU and Stereo cameras tomorrow and see how well Rovio works with my configuration.

closed 2020-02-03 by huaizheng