rpng/R-VIO #10: About feedback from mapping

canyou2014, 2019-07-29:
I have read this paper https://arxiv.org/abs/1903.08636,  it is about feedback map point to MSCKF by Schmidt-EKF. If I wan to realize Schmidt-EKF on  R-VIO，How should I propagate the covariance matrix  PAS?

goldbattle, 2019-07-29:
Thanks for being interested in both of those projects.
You can propagate the P_as using equation (7) in that [paper](https://arxiv.org/pdf/1903.08636.pdf):
[image]

An important note is that the schmidt states do not evolve with time as they are represented in the global frame of reference. If you represent your SLAM features in the robot centric frame then you lose the computational advantage as you need to perform the "composition" to warp the features after each update. You should look more carefully into this part.

huaizheng, 2019-07-29:
Hi @canyou2014, for such extended problem I recommend you to directly email either me or @goldbattle for discussion, which I think it is better than opening an issue. Thank you.

closed 2019-07-29 by huaizheng