rpng/R-VIO #19: How to use this code to publish a odom node?

sunt40, 2020-07-09:


huaizheng, 2020-07-09:
For this, you can instead fill the output of estimator, q and p, into a new 'nav_msgs::Odometry' message and then publish it, based on the current codebase. You can find an example here [http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom](url).

closed 2020-07-10 by huaizheng

sunt40, 2020-07-18:
I have read your code but I have no idea how to modify your code to public the /odom topic. Can you give me some further advices? Thank you.

huaizheng, 2020-07-18:
Please check my code in rvio/System.cc, where I publish two types of messages at the end, tf and nav_msgs::Path, once having the optimal state estimate. So if you would like to publish other messages, you can put your code there referring to the example in my last comment.

sunt40, 2020-07-18:
The following is what I have added code:
`#include <nav_msgs/Odometry.h>
#include<tf/transform_broadcaster.h>

nav_msgs::Odometry odom;

mOdomPub = mSystemNode.advertise<nav_msgs::Odometry>("/odom", 0);

//broadcast odom 
    geometry_msgs::Odometry odom;
    //geometry_msgs::Quaternion odom_quat = tf::createQuaternionMsgFromYaw(th);
    tf::TransformBroadcaster odom_broadcaster;
    odom_broadcaster.sendTransform(transformStamped);
    odom.header.stamp = current_time;
    odom.header.frame_id = "odom";
    
    odom.pose.pose.position.x = pGk(0);
    odom.pose.pose.position.y = pGk(1);
    odom.pose.pose.position.z = pGk(2);
    odom.pose.pose.orientation = qkG(0);
    
    odom.child_frame_id = "base_link";
    odom.twist.twist.linear.x = xv;
    odom.twist.twist.linear.y = yv;
    odom.twist.twist.linear.z = zv;
    mOdompub.publish(odom);`

Obviously,there is a lot of mistakes,so please give me your advice. @huaizheng

sunt40, 2020-07-19:
And, I don't really understand the meaning of many of your parameters. This makes it very difficult to write  code.

huaizheng, 2020-07-19:
First, if you don't understand the meaning of parameters, the best way is to go back to read the papers where all the equations have been released, and this code is the exact implementation. Also, I have given you my advice about your problem, just a ROS-related practice, which definitely should not be difficult to solve by yourself.

sunt40, 2020-07-19:
OK。I have download your paper and I will read it carefully.

sunt40, 2020-07-21:
I  am a beginner of VIO and your algorithm is quite novle. Now although I can publish a "odom" topic,I tried to use some parameters but still failed to publish the correct odometry messages. The following image is the details of the "odom" publish. Can you tell me which parameters I should use in your code ? Thank you very much! @huaizheng

huaizheng, 2020-07-21:
I just implemented it myself and have pushed the code, please go and check how I made this work.