rpng/R-VIO #21: About the algorithm‘s harmful and unbelievable drifting when it worked in several static situations below

muchen12, 2020-10-06:
Hello, I am sorry to bother you. I want to seek some help. Well, there are several confusing questions we met as follows,
(1)When followed those instruction from file 'Readme.md', we ran this algorithm with Euroc datasets, such as MH_01_easy.bag, we suffered a really tough situation that the trajectory can always drift far away with an incredible rate at beginning, especially when the mav was undergoing a long time static beginning. This directly resulted in the whole process failed. But in your paper, I did not find the same problem when you ran this algorithm under these same datasets. Could you please give me some advice? Is this a problem from code? Or something else possible.
(2)It is similar to (1) that when mav keep short time static during process, such as some periods in MH_01_easy.bag, although the trajectory has been failed because of the static beginning, I noticed that the trajectory was still extending with a fast speed. Meanwhile, for example, I noticed the qkG was keeping increasing. In paper, it claims that the algorithm can handle such situation, but why does the estimation process become so bad?
That is all. I really hope someone can answer our questions. Thanks a lot!

huaizheng, 2020-10-06:
Sure, I am glad to answer your questions. First of all, R-VIO is a pure odometry system without building a map, although you can find some other VIO systems which build map. At the beginning of dataset there is no feature constraint can be used for pose estimation as the feature triangulation cannot be performed during stationary phase, thus only IMU measurements can be used for pose estimation. However, as we know that IMU measurements are corrupted by high-frequency noises, the corresponding pose estimate can easily drift if directly integrating those IMU measurements, which is just what you experienced.
1) Basically, the euroc dataset is designed for general SLAM case, so the up-and-down stage at the beginning of each MH sequence is primarily used by SLAM system to initialize a map before actually moving out. While, this stage is useless for R-VIO because it can start after a fast and simple initialization, as shown in our paper, thanks to the proposed robocentric formulation. Therefore, when I tested on MH sequences, I skipped this stage and started R-VIO around the timestamp where the mav was about to take off for exploration. Also, a static detection is realized in the code simply by judging on some thresholds, however, if you have any better idea you can implement it yourself.
2) It should be noted that, the mav was actually not totally stationary during the stage you mentioned as its rotors were always working. This would easily confuse the static detection of R-VIO, thus I have the same suggestion as above. It should also be noted that, in our paper we emphasize that with our proposed robocentric formulation the VINS system has better observability properties under the special motions. This will facilitate the real system to deal with those situations, and we have shown that in the urban driving case it could mitigate the scale drift when the car stopped. However, as the real-world condition may not entirely follow the theoretical base, namely zero-mean white Gaussian noise assumption, we should not simply expect it to solve the problem for all the cases.

muchen12, 2020-10-07:
> Sure, I am glad to answer your questions. First of all, R-VIO is a pure odometry system without building a map, although you can find some other VIO systems which build map. At the beginning of [...]

muchen12, 2020-10-07:
Oh, thank you very much! Really thanks.
Morning, I retried this process with your valued instruction 'skipped this stage and started R-VIO around the timestamp where the mav was about to take off for exploration'. I used V1_02_medium.bag from 10 seconds after bag playing, which is just the moment you mentioned above. But the trajectory still drifted far away with a fast speed. The trajectory produced did not look anything like the real exploration. As the picture showed below, it seems that visual information did not give some help to  restrain the IMU's high drifting. But from the tracking phenomenon showed in picture, with the consideration of your codes in track.cpp, the distinguishment between inliers and ouliers exactly received the 'R' constraint from IMU. So, this is really confusing me. Manito, how can I do to achieve those normal and excellent trajectory charts showed in your paper? Least for this V1_02_medium.bag......
(Uh...I have tried my best but still do not know how can I do to reproduce those good trajectory results in your paper.I haven't made any changes to your code. So please help me. I do not sure what went wrong. Thank you very much.)
[image]
[image]

huaizheng, 2020-10-07:
For the V* sequences you don't need to skip the beginning part, because there was no up-and-down phase and the mav stayed there more stably than in the MH_* sequences such that the static detection is able to provide reasonable judgement. So you can just play the rosbags of V* from the very beginning.

muchen12, 2020-10-07:
Well, Dear Manito. To be honest, actually, I have also tried in this way for many times before I consulted you for some advice yesterday. But till now, none of them have been successful in getting a trajectory that corresponds to the actual motion, because of this pathological and violent drift. I haven't made any changes to your code. What I have done is just 'git clone' and do those thing followed what you say in 'Readme.md'. I don't think this is a problem with your code, but I'd like you to give me a few tips on how to fix this situation. Thank you~

huaizheng, 2020-10-07:
Currently I have no clues based on your description, however, this is just what I get on this sequence and I did not find any issue.
[image]

muchen12, 2020-10-07:
I have found out the reason why it always caused serious drift. Thank you very much for your patient response. I'm so sorry to have bothered you so long. It is those ROS messages out of step that cause the harmful drift. 
Thanks again. Thank you very much~

closed 2020-10-07 by huaizheng

Gatsby23, 2021-11-13:
> I have found out the reason why it always caused serious drift. Thank you very much for your patient response. I'm so sorry to have bothered you so long. It is those ROS messages out of step that cause the harmful drift. Thanks again. Thank you very much~
Sorry for disturbing you. But I have met the same problem as you are. What is the reason to caused this serious drift?