rpng/R-VIO #22: Please excuse me. Some questions about algorithmic equation deducing and a certain experiment operation details.

muchen12, 2020-10-11:
Please excuse me. I am sorry to bother you again. Since the last time I consulted here, I found the reason why I failed to run V* and M* Euroc datasets. But unfortunately, I still met some problem, so that I want to ask for some details about the process you ran it. Besides, I also have a algorithmic equation deducing question about your IJRR paper 'Robocentric visual–inertial odometry'.
(1)My algorithmic equation deducing question is around equation (42)(43) as picture showed below. (Especially middle part 10~12 columns)
[image]
I do not know how equation (43) was achieved. So if possible, could you please show some necessary derivations? Thank you very much.
(2)Next  is that I want to consult what kind of methods about bag playing you adopted when you accomplish the complete experiment proc, especially when you did those algorithmic testing. Last time, I realized the reason why I suffered from failure when I tried to run "rosbag playing" command. It is the IMU and images'  topic message out of sync that resulted in morbid drift. So I tried to keep pressing "s" button after I ran "rosbag playing" command. And I obtained same trajectory results as you got. But it brought a new problem here. If I did it in this way, the speed of bag playing will be too slow. Because of this, I could not try those test about running time. I have tried storing IMU and images datas into two buffer queues, something like the way in VINS-mono. But it still result in the same morbid drift phenomenon as I directly ran  "rosbag playing" command without keeping 's' button pressing. So it is the reason why I want to consult what kind of methods about bag playing you adopted when you finished your paper. Thanks very much.

muchen12, 2020-10-17:
Is there anyone that can help me please? I have pushed this issue here for almost a whole week. Could you please help me?

huaizheng, 2020-10-17:
1) In our robocentric formulation, the frame of reference needs to be changed from one IMU frame to the next which is achieved through the composition step. In addition to the change of reference for the state (40), the covariance matrix should also be expressed in the new frame of reference. This is done by the so-called covariance propagation (41) with a corresponding Jacobian matrix (43). To derive it, you should start from the original composition equation whose form is the same as (40), however, now without "^hat" for each state. And for the relevant procedures you can refer to our companion tech report (please check the reference) where we have provided a lot of similar derivations for the other equations shown in this paper to get their Jacobians, which can help you to derive (43) yourself.
2) As R-VIO dose not need "SLAM wiggle" for initialization, you can skip all those stages in the dataset and then start it. This also works if you don't want to wait for a long stationary period. For example, the mav takes off after around 11 seconds in V1_02, so you can first play rosbag for 8 seconds and stop (with --pause option), then roslaunch R-VIO, and finally resume playing rosbag to let the estimation begin. I just tried this myself before writing this comment and it also worked, and the result was just the same as what I showed you before. Actually, when you enable --pause option rosbag play will preload dataset so that you don't need to worry about the timestamp issue, because it just replays all the messages included. So this is why I have no clue for the issue you met. If such thing occurs all the time, I may recommend you to try it on another computer to see what result you can get.

closed 2020-11-03 by huaizheng