rpng/R-VIO #27: Question About the F matrix in code.

Gatsby23, 2021-10-20:
Hi, The paper shows that the F Matrix is like this:
[image]
But the code is multiplied with the mnGravity. What does it meaning?
[image]

huaizheng, 2021-10-20:
In the implementation, we treat local gravity as a vector with constant magnitude (e.g., 9.8), therefore, we actually estimate a unit (directional) vector of local gravity.

huaizheng, 2021-10-20:
There are the following differences between Rovio and R-VIO: 1) Rovio uses raw IMU measurements to directly update the pose of global frame, while R-VIO uses relative motion to incrementally update the pose of global frames (i.e., using the composition); 2) Rovio implements an EKFSLAM-based formulation (i.e., jointly estimates the landmarks), while R-VIO implements an MSCKF-based formulation (i.e., pose-only VIO); and most importantly 3) R-VIO has been proved to have invariant observability properties (i.e., its unobservable subspace is constant) which significantly improves the estimation consistency under generic motions.

Gatsby23, 2021-10-20:
> There are the following differences between Rovio and R-VIO: 1) Rovio uses raw IMU measurements to directly update the pose of global frame, while R-VIO uses relative motion to incrementally update the pose of global frames (i.e., through the composition); 2) Rovio implements an EKFSLAM-based formulation (i.e., jointly estimates the landmarks), while R-VIO implements an MSCKF-based formulation (i.e., pose-only VIO); and most importantly 3) R-VIO has been proved to have invariant observability properties (i.e., its unobservable subspace is constant) which significantly improves the estimation consistency under generic motion.

Thank you very much~
真是特别好的工作，我将再仔细看看代码和论文，体会当中具体的不同之处