rpng/R-VIO #32: Question about the update.

Gatsby23, 2021-11-09:
[image]
Sorry to disturbing you. There are some question in the updating process. In Updating process, the error state about the global variable should be 0, right? But why there are still the update correction in line 549？
不好意思，大神。关于更新过程，这里还是有地方不懂想请教您一下。关于状态量中的全局变量，似乎在更新的时候一直为0?但为什么在程序(Updater.cc)第549行依然对它有误差修正呢？
另外有个简单问题是，为什么这里的角度误差量都要乘以0.5呢？我看Quaternion ESKF的代码中，这种误差状态似乎都是小角度更新直接相乘？不会考虑二倍关系？

huaizheng, 2021-11-09:
The global state is jointly estimated along with the other states. Though it is not used to form the measurement model, its covariance with the other states make it possible to be updated every time, therefore, the value of its error state should not be 0.

Gatsby23, 2021-11-11:
> The global state is jointly estimated along with the other states. Though it is not used to form the measurement model, its covariance with the other states make it possible to be updated every time, therefore, the value of its error state should not be 0.

Thank you for your quickly reply, Thank you very much!

closed 2021-11-11 by Gatsby23