rpng/R-VIO #35: Question about the trick R-VIO used in the real environment.

Gatsby23, 2022-01-19:
Dear Researcher:
 I'm a Phd student from Shanghai Jiao Tong University. I have read the paper《Robocentric Visual-Inertial Odometry》recently. I really appreciate the idea that uses robocentric formulation to avoid the inconsistent problem and can avoid the initialization failure. With curiosity, I have test the algorithm under different environments. The simulation gazebo environment and the real indoor experiment environment.   I have noticed that in the gazebo environment, the algorithm can show a better result than others however, in the real environment, the R-VIO often get failures. With grateful and thanks, I want to know is there some tricks to use this algorithm in the real environment? Here is my consideration:

-  The extrinsic parameter between IMU and Camera, the R-VIO uses JPL formulation. So the extrinsic should represent the camera to the imu.
- The imu intrinsic parameter(ba or bg) should be little larger to include some additive noise, like #23 
- The initialization module should be carefully set the detection threshold. The threshold have a deep impact on the bias and gravity estimated. 

Here is my tips about the R-VIO used in the real environment. However, from your practice. Is there are some other useful information that I have missed ? Could you please give me some good advices about how to tune the code suitable in some uav environment?  If I re-implent this algorithm, what else should I be careful about?

Thanks
Xinjiang Wang

huaizheng, 2022-01-19:
Thanks for your interest in this work @Gatsby23. This code is a proof-of-concept for our new formulation, and we believe there is space to improve further. Currently for real experiments, we recommend two things: 1) accurate spatial-temporal calibration parameters, and 2) starting from a stationary pose, because this code does not include online calibration and dynamic initialization.

It should be noted that: 1) changing the format (from rotation matrix to JPL (or Hamilton) quaternion) does not change the physical meaning of rotation, therefore, the order of frames should be the same (Using "from one frame to another" to explain R_ic can be quite confused ({i} to {c}, or {c} to {i}?), instead I would like to say that R_ic (or {^i_c}R in latex) is used to transform a state in frame {c} to frame {i}). 2) Manually changing the value of sensor noise parameter may not be a good practice, because this suggests that the parameter value for use is actually not accurate. 3) The initialization in this code performs well if starting from a stationary pose, because this favors a good initial estimate for bias.

It would be a good start from the EuRoC dataset if you want to do some uav applications. However, it would become challenging because the dynamic scenario including the illumination change, sensor parameter change and the disturbance from moving objects can easily degrade the performance of a system, which should be taken care of.

Gatsby23, 2022-01-20:
> Thanks for your interest in this work @Gatsby23. This code is a proof-of-concept for our new formulation, and we believe there is space to improve further. Currently for real experiments, we recommend two things: 1) accurate spatial-temporal calibration parameters, and 2) starting from a stationary pose, because this code does not include online calibration or dynamic initialization.
> 
> It should be noted that: 1) changing the format (from rotation matrix to JPL or Hamilton quaternion) does not alter the mathematical meaning of rotation, therefore, the order of frames should be the same (Using "one frame to another" to explain R_ic can be quite confusing ({i} to {c}, or {c} to {i}?), instead I would like to say that R_ic (or {^i_c}R in latex) is used to transform a state in frame {c} to frame {i}). 2) Manually changing the value of sensor noise parameter may not be a good practice, because this suggests that the parameter value for use is actually not accurate. 3) The initialization in this code performs well if starting from a stationary pose, because this favors a good initial estimate of bias.
> 
> It would be a good start from the EuRoC dataset if you want to do some uav applications. However, it would become challenging because the dynamic scenario including the illumination change, sensor parameter change and the disturbance from the moving objects can easily degrade the performance of a system, which should be taken care of.

Thanks for your quickly replay, I'll read the advice more carefully. Futhermore, is there some paper about dynamic initialization?

closed 2022-04-06 by Gatsby23